Sinapse-Energia/bootloader-stm32 #2: Testing firmware downloading with GPRS layer transport.

francisjjp, 2017-06-30:
It was watched following things:

1) The searching of firmware file from bootloader side is done using a complete firmware name. i.e: CMC.bin. It means that if all CMC have same bootloader program, all CMC are going to download the same program from server. There will be that modify the 'GET' command sent to http server in order to avoid downloading always same program. i.e: The bootloader program should search in some position in one shared memory sector with application program one specific directory name o one name. If we want to use same bootloader program, we must include that bootloader program search in one shared memory sector with application program, or searching in one only ID  saved in each microcontroller. If is not possible, there will be as many bootloader programs as aplication programs.

2) Downloading progress using actual GPRS layer transport is slow. Proccess of connection is slow. Each retry takes 1minute. If there is at least 5 retries, the overall process can take 5 minutes plus downloading process. After this, the application program must connect to broker (1 minute more). In both cases, if there is a lot problems to connect with network, GPRS process becomes slower because of recommended connecting process (quectel recommends that if there is problem of connectivity, wait 180 seconds). All this proccess is done in each restarting!!. This must be improved asap. 

3) There is one issue with application CMC-Easynet. It freezes suddenly at starting.

francisjjp, 2017-06-30:
4. If CMC.bin from remote server is erased. Then, at 0x8008000 position in flash memory is filled with one HTML response from server and not any program.

francisjjp, 2017-06-30:
5. It is needed that bootloader has IWDG activated in order to avoid get freezes under bootloader program and it do not work anymore.

francisjjp, 2017-06-30:
6. Because of failure of downloading a HTML response, no more application programs are downloaded. It is needed to erase with STLink complete chip, program the bootloader program, and let in remote http server the file CMC.bin

sinapseenergia, 2017-07-04:
Testing Sheet for CMC integration

[CMC Bootloader Integration Testing Sheet.docx](https://github.com/Sinapse-Energia/bootloader-stm32/files/1121962/CMC.Bootloader.Integration.Testing.Sheet.docx)