Sinapse-Energia/bootloader-stm32 #3: [Shared memory] To define a part of shared memory that contains all the necessary data to be shared by the bootloader and the application

ralcaide, 2017-07-02:
The bootloader and the client application should share some information in order to work properly. 
It is necessary to define the where and what should be shared. As a minimum, the following data should be accessible by both applications:

- DeviceID : Int / Set in the fabrication process / Bootloader read it to create the downloading URL / Client application read it to implement several features  
- Relay status: Array of 7 boolean / Initially set to false / Bootloader read it to maintain the device status during the updating process / Client application read and write it in order to implement several features

sinapseenergia, 2017-07-03:
This part depends on device.

For CMC , described data are enough.
CMC case, we can use external eeprom to store shared data, but it maybe slow for our needed features.

Not all devices will have EPPROM, but most of device will  use STM32 microcontrollers

STM32 has a good feature to store data safetly and quicly and easier than flash

"Battery backed RAM" using Vbat (Which will be used too for RTC)

_"The obvious alternative, without adding an external EEPROM, is use battery backed RAM. This is slightly more complex to program than using an in-memory variable because it is part of the RTC which needs to be initialised. However once it set up, it can be read and written with only a few extra instructions. So it's access time is sub-microsecond, fast enough to use in an interrupt service routine, and it won't wear out.

This does mean you need to be able to get at the VBat pin, and connect it to a small battery, e.g. a hearing aid battery"_

Maybe we have to consider this (Francis, please, analize this option)

sinapseenergia, 2017-07-03:
More information about battery backed SRAM:

_After reset, the backup domain (… backup SRAM) is protected against possible unwanted write accesses. To enable access to the backup domain, proceed as follows …_

_It gives you instructions on how to gain access to the backup domain by directly writing to the certain peripheral register. If you have access to the STM32F4xx library, you can call something like this (note: I'm using the STM32F2xx library):_

**PWR_BackupAccessCmd(ENABLE);**

_Note: There's is more to it than simply calling the above function, such as enabling the backup SRAM interface clock. Consult the STM32F4 series documentation._

_There is a lot of documentation embedded in the library source that is invaluable and if it's available should be read._

_On the STM32F2 series microcontroller, SRAM is located at the following memory address range:_

    **0x40024000 - 0x40024FFF**

_And can be written to somewhere at location, for example, as follows:_

**#define VAR_LOC ((volatile uint8_t *)(0x40024000))
volatile uint8_t *pVar = VAR_LOC;
*pVar = 5;**

sinapseenergia, 2017-07-03:
More info avaiable here:

https://stackoverflow.com/questions/20667754/how-to-use-backup-sram-as-eeprom-in-stm32f4