Sinapse-Energia/bootloader-stm32 #4: [Initial status] Set the device in a known initial state

ralcaide, 2017-07-02:
Before start with the update process, the bootloader should set the device in an initial and known state. This state is got from the value of the variables accessible through the shared memory. See #3

This function should do the following:

- Read the status of the relays
- Set the relays to the read status (ON / OFF)

sinapseenergia, 2017-07-03:
This fuction have to be very quick in order to avoid glitches, so it´s mandatory to use the optimal features micro to permform it.

francisjjp, 2017-07-04:
There is one strange issue testing behaviour. In bootloader mode is detected following values for first 4 relays in one CMC: 0, 0, 0, 1. In application client values are: 1, 1, 1, 1. ???