Sinapse-Energia/bootloader-stm32 #11: [Integration - WIFI] To integrate the bootloader working with ETH AP devices

ralcaide, 2017-09-24:
It is necessary to integrate the bootloader into AP devices working with ETH / WIFI. To do that is necessary to modify the wifi driver in order to work with AT Commands.

fpgamcu, 2017-09-27:
Here is patch with AT integration, should I push it to git, or leave as patch?
[0001-Modified-WiFi-driver-added-AT-commands-https-github..patch.txt](https://github.com/Sinapse-Energia/bootloader-stm32/files/1336955/0001-Modified-WiFi-driver-added-AT-commands-https-github.patch.txt)

francisjjp, 2017-09-27:
@fpgamcu .

Regarding this issue. Have you tested that some example program is loaded using wifi?.

fpgamcu, 2017-09-27:
I tested that AT commands is changing server name in `Socket_Init()` function and connects to those server, also tested that bootloader successfully execute code from its address.
Also for enable WiFi to work I think you need to change line 14 in Definitions.h from `#define LOG_WIFI 1` to `#define LOG_WIFI 0`. And also to uncomment block at line 125 in Bootloader.c and comment block at line 141.
When I made this changes and execute bootloader I get "BOOT ERROR No length FW valid" using current server.

fpgamcu, 2017-09-28:
Hello, I added define for transparent mode and performed small cleanup.
This code tested with ETH connection and local HTTP server, working good at my board