Sinapse-Energia/bootloader-stm32 #13: [Migration to F2 family] To migrate the bootloader to F2 family

ralcaide, 2017-11-15:
Now, we have two versions of the bootloader: F4 and F0.

It is necessary to have also the version for the F2 family. We don't know if should work like the F0 (Remember the use of remap() fuction) or like F4 (interrupt vector is relocatable)

To create a new branch with the F2 code.

francisjjp, 2017-11-28:
This branch NOT WORK YET. There is some bizarre issue with cproject or debugging. (I suppose). I can not even to blink a led!

ralcaide, 2017-11-28:
Why do you think is a problem with .cproject? You already have a valid .cproject for M3 in the client application, maybe is useful for you in order to get the problems

ralcaide, 2017-11-28:
In the commits history we see a lot of commits trying to fix HFs but without fixing the problem, actually. Why do you commit that?
Maybe is better to go back to the last "right" commit https://github.com/Sinapse-Energia/bootloader-stm32/commit/9346ac3572348fd9c1bf0de1857a47e6df101741 and start again with the modification of the .cproject

francisjjp, 2017-11-29:
I can debug now. Program is loaded with CRC ok, but it seems there is problem with loading interrupt vector. App client starts ok with switching on the green LED but it never connects to broker.

francisjjp, 2017-11-29:
OK. App program is loaded and executed right. It is needed to include status with led for indicating that we are in bootloader mode.

ralcaide, 2017-12-04:
Integration with new version of GPRS is still not working. The buffer used by the Bootloader is different to the EC-M2M. It is necessary to adapt the Sekas code in order to use the new Buffer (DataBuffer)

francisjjp, 2017-12-05:
@ralcaide. @sinapseenergia .

Now, NRTRIES is only 1. Only one commutation of relay is done if needed, not more. If aplication program BIN does not exist in server, round 15-20 seconds is the time that bootloader takes before application client runs. One blinking red with 500ms ON - 500ms OFF in one infinite loop is done if there is not exist one preexistent application program in memory, and it is not possible to download the application from server. Then, user must switchoff-switchon the device.

ralcaide, 2018-01-15:
Done. Closing the issue . Tag: https://github.com/Sinapse-Energia/bootloader-stm32/releases/tag/M3_v1.0

closed 2018-01-15 by ralcaide