Sinapse-Energia/bootloader-stm32 #14: [Bug]. Problem with STM32F405VG and bootloader cortexM4

francisjjp, 2017-11-22:
There is now issues with bootloader and ONLY GPRS with cortexM4 branch.

The starting point is last commit: 5735811733d07f31d388e064b0d07ad55f47857b

Problems detected:

1) Some times M95_OK connection is never reach. If so. It is needed to use retries variable with value >=3 instead 1.
2) Problem is at function: **BOOT_ERRORS Boot_PerformFirmwareUpdate(void)**

```
...
...
...
// Clear buffer flash first
     FlashNVM_EraseBank(FLASH_BANK_COPY);

    // Get data
    sprintf(boot_buff, "GET /%s HTTP/1.1\r\nHost: %s\r\n\r\n", HTTP_SERVER_FW_FILENAME, HTTP_SERVER_IP);
    Socket_Clear(ssource);
    Socket_Write(ssource, boot_buff, strlen(boot_buff));
    // Read answer
    Socket_ClearTimeout(ssource);
    total_len = 0;
    while (!Socket_GetTimeout(ssource)) {

        len = Socket_Read(ssource, boot_buff, BOOT_BUFFER_SIZE);
        if (len) {
        	if (WDT_ENABLED==1)  HAL_IWDG_Refresh(&hiwdg);
        	FlashNVM_Write(fl_addr, (uint8_t*)boot_buff, len);
        	total_len += len;
        	fl_addr += len;
        	Socket_ClearTimeout(ssource);
        }
    }

    if (total_len < 10) {
    	//No file on server!
    	if (LOG_WIFI==1) HAL_UART_Transmit(&huart6, "(BOOT ERROR No file on server)\r\n", 32,100); //Francis, for logging
    	return BOOT_ERR_CONNECTION; //Err
    }
....
...
....

```

total_len is always = 9. There is no any response from server, even no 404 response and firmware always exists.

We have tested over sinapseenergia.com:80 server and CMC_FRAN.bin with crc ok

**There is into external one DUT with APN=im2m.matooma.com. User =movistar Pass = movistar.**

ralcaide, 2017-11-23:
@francisjjp , I know that there is a maximum size for the executable. Have you checked that? Could you try with an AP binary that we know that the size fits?

ralcaide, 2017-11-23:
At this moment, Sviatoslav is totally focused on AP and CMC and Sergey does not have other resource available. We are waiting for feedback