Sinapse-Energia/bootloader-stm32 #15: [General] Bootloader-Client Shared Capabilities (UpdateFirware and DecrementalCounter)

sinapseenergia, 2017-11-30:
Regarding last update, a contingency mechanism is required in case Client FW fails.
If Client has the control to set if the bootloader download and update FW, we have to add a second condition in case client is not working well.

So, it´s neccesary to add a boolean var (UpdateFirmware) and a numerical counter N.
Client set this counter to a Number, and Bootloader decrement this number each time it runs.
In case client is corrupted, N finally will be 0 and then bootloader will try to download and update Client FW

So resuming:
Shared vars regarding this:
Bool UpdateFirmware
Int DecrementalCounter

Client Set:
UpdateFirware --> Under User Demand
DecrementalCounter --> Set it to a number (i.e. 10)

Bootloader will run each restart, and decrease (if >=1) DecrementalCounter then check

If UpdateFirware==1 or DecrementalCounter==0 --> Download and Update

else--> go to client

THIS CHANGE HAVE TO BE APPLIED TO ALL FAMILIES / FW CLIENTS SO WE HAVE TO CHECK THIS SPECIFICALLY

ralcaide, 2017-11-30:
@francisjjp, to develop that take into account that:
- Get the variables from the shared memory with the client
- Now, these variables don't exist
- So, by default, if these variables don't exist use a "secure default value" that allow us to update the FW.  for example (true, 10). 
- Then, we will update the client and we will have a mechanism to set the UpdateFirmware to 0

francisjjp, 2017-11-30:
OK, I can simulate this way. Later I need following:

UpdateFirmware.-> Start address in flash and number of bytes.
DecrementalCounter-> Start address in flash and number of bytes.
Relay1Status-> Start address in flash and number of bytes.
PWMStatus -> Start address in flash and number of bytes.
nameFileBINtoDownload -> Start address in flash and number of bytes.
APN -> Start address in flash and number of bytes.

Did I forget another else?

francisjjp, 2017-11-30:
UpdateServer -> place where BIN is downloaded from

francisjjp, 2017-12-03:
Some comments:

- I can not include the PWM setting in bootloader mode, because size of bootloader is increased and then take first sector of application program (bootloader is higher than 32kb, Only I can set the relay). Either we try to use register setting without using HAL libraries, or we increased sector for bootloader, then we must put one offset to APLICATION_PROGRAM. Another possibility is not to control the PWM in bootloader mode.

- We must define watch if there is one possibility to lost device. Basic testing has been done and it works.

francisjjp, 2017-12-03:
Another comment: @ralcaide , @sinapseenergia 

Now, app client program only puts to 9 the UPDFW_COUNT variable when devices is connected to internet.

There is one good thing and one bad thing with this:

a) Good thing: The mode to enter in bootloader mode even if there is not one MQTT command for this is to switch off-switch on the devices 8 times consecutive before devices connects to internet. So, red led  is switching on and we have entered in bootloader mode.

b) If there is problem with internet connectivity, at 8 times that devices restarts, it enters in bootloader mode.

If we set the 9 value UPDFW_COUNT immediately at starting of application client, we have no possibility to update a new firmware if MQTT command for activating bootloader mode does not exist.

ralcaide, 2017-12-03:
@francisjjp , several things / questions:

1. Size: We need to integrate the new version of GPRS because maybe we need more space or mayble less... It is important to be sure that the program with new GPRS fits in the memory. I think, we have more memory in this device so could be a good idea to increase the memory available for the bootloader, isn't?

1.1 I think, the PWM is necessary, so, please increase the memory available for the bootloader and commit the changes. This will be the stable version with the old GPRS

2. I think the behavior of UPDFW_COUNT is right. We will develop the command to set UPDFW to true asap. With UPDFW = true, we will enter always in bootloader mode.