Sinapse-Energia/bootloader-stm32 #18: [Integrate] Integrate bootloader for EasyNet v1p0 Board

sinapseenergia, 2017-12-12:
Integrate code into EasyNet v1p0 Board

Changes regarding CMC Commboard are just about Microcoller pinout. Adapt it and create a different Branch

ralcaide, 2017-12-12:
This issue is not immediate:
- The branch should be created from the branch Cortex M4: https://github.com/Sinapse-Energia/bootloader-stm32/commits/CortexM4
- The last version of this branch was not working in GPRS. At first is necessary to test that works with ETH
- Then perform the migration to the new commboard
- Sva has included several changes into the USR management that should be also included into the bootloader
- Here are all the modifications performed during the migration to the new commboard: https://github.com/Sinapse-Energia/AP-Sinapse/commits/Commboard-V2.0
Commits related with the issues 76, 77 and 78

francisjjp, 2017-12-13:
Where is the branch to use for this development?, can you create it?.

ralcaide, 2018-01-15:
This integration is ongoing by Sviatoslav following other issues. Closing this issue because the development is managed in other way

closed 2018-01-15 by ralcaide