Sinapse-Energia/bootloader-stm32 #24: [Commboard V1p1 - M4 GPRS & ETH] To manage both interfaces in parallel

ralcaide, 2018-01-15:
SIM_DETECT--> YES --> TEST ETH --> Yes -> Use ETH
SIM_DETECT--> YES --> TEST ETH --> NO -> Use GPRS
SIM_DETECT --> NO -> Use ETH

fpgamcu, 2018-02-13:
Now in code implemented next interface management
TEST ETH --> Yes -> Use ETH
TEST ETH --> NO --> SIM_DETECT --> YES -> Use GPRS
TEST ETH --> NO --> SIM_DETECT --> NO -> Load firmware from flash
Is it correct?

ralcaide, 2018-02-13:
Yes, it is correct. If there is not connectivity, then launch the client application already installed in the flash. If there is no application, the bootloader is executed again...

ralcaide, 2018-03-06:
Tested in V1p1 with F427VG client program and is not working:

- ETH : Finish with HF
- GPRS : The program is blocked

The WDT is disabled.

@fpgamcu, could you please take a look on that. We need ASAP a bootloader working with the two communication interfaces in V1p1 with F427VG.

ralcaide, 2018-03-07:
There was a misunderstanding about the HW characteristics:

We have several kind of devices and we don't mount all the HW in each device. There is always a communication interface but not both

the Bootloader should be compatible with the three possibilities. ETH, GPRS, BOTH

ralcaide, 2018-03-07:
External 1 tested (ETH) : OK