Sinapse-Energia/bootloader-stm32 #26: [Cortex M3 - GPRS Engine] To integrate last version of GPRS engine

ralcaide, 2018-02-26:
The actual GPRS engine is only working with M95. To integrate the last version developed in the M2M project that is able to work with M95 and BG96 transceivers

juanra-bascones, 2018-02-26:
Is this project the Bootloader?
Is to integrate the Transceiver component into it?
Where do I have to download  the project from?

ralcaide, 2018-02-26:
Yes, you should download this repo and integrate the last version of the GPRS engine in the branch Cortex M3. There is already a version integrated, so it should be easy to integrate the last changes. 
Here you can see history: https://github.com/Sinapse-Energia/bootloader-stm32/commits/CortexM3

ralcaide, 2018-02-27:
It depends, this is your local repository. Do you have something important in the BL? I suppose not but maybe you could only to update this repo if it is the Bootloader repo.
Two options:
- Update the existing repo (git pull)
- Create a new one with other name or with the same name after delete the previous one

juanra-bascones, 2018-02-27:
OK
I renamed the old (just ion case..) and download the new with the very same name