Sinapse-Energia/bootloader-stm32 #28: [Cortex M4 - Commboard V1p1] GPRS is not working

ralcaide, 2018-02-26:
GPRS bootloader in Commboard V1p1 (New) is not working. Please, fix it

ralcaide, 2018-02-27:
Great, where are you tested that? external or your device? with what server?

fpgamcu, 2018-02-27:
On my device with my server. I can also test on external. What server would be better to use and on what external?

ralcaide, 2018-03-15:
The bootloader for GPRS in V1P2 is NOT working - (External 2). The client program is working without problems.

It is received a HF  here:

if (cmd->reply.handler){
                        int rc = cmd->reply.handler(dest);

line 376 of M95lite.c

@fpgamcu , could you take a look, please?

ralcaide, 2018-03-18:
@fpgamcu , could you provide feedback about the above problem?

fpgamcu, 2018-03-19:
This morning I was not able to debug the bootloader on External 2, please see attached video, it stuck at main code start, maybe problems with debugger
https://www.dropbox.com/s/rbwa5tjrj6ovytk/Debug%20problems.ogv?dl=0
On my device I can not reproduce the problem.

ralcaide, 2018-03-19:
Ok thanks, yes, this problem appears yesterday. We will try to fix it asap and I will let you know

ralcaide, 2018-03-19:
Tested: Works after several tries. 
To be tested more in deep

ralcaide, 2018-03-20:
Tested during the night.

27 Tries - > 23 OK

We can assume the bootloader is working as it is right now. With several limitations but working