Sinapse-Energia/bootloader-stm32 #31: [Cortex M4 & M3] To include HTTPS as a protocol for downloading

ralcaide, 2018-04-03:
To develop HTTPS as a protocol for connection to the server and downloading files

ralcaide, 2018-04-03:
@fpgamcu , could you give an estimation about to do that in Cortex M4?

@juanra-bascones , could you give an estimation about to do that in Cortex M3?

juanra-bascones, 2018-04-03:
OK.
At first glance, I can see  TWO possible alternatives:
1) Try to simply migrate the flow for secure connection into the bootloader, and then, keep the HTTP iteration loop just as is. 
2) Try to explore the transceivers capabilities to support internally both, HTTP and HTTPS protocols. 

If the first gets right to the point, is, for sure, the fastest and the one with less impact in the current program structure, and, also the one with less dependence on the transceiver.
(simply, I am not 100% sure about HTTPS is nothing more than HTTP verbs over TLS or there is something more unther the hood..).
So, I will go a little deeper on that, and also I will try to make some kind of experiment with the EVBs..

fpgamcu, 2018-04-03:
I have worked with HTTPS before, but in opposite task, implementing HTTPS web server on STM32F7.
Indeed, HTTPS have the same protocol as HTTP, but all data transfer is encrypted. I used mbedTLS for encryption. We will need time source in order to have today date and manage certificate expired date. Also mbedTLS library can take significant amount of flash memory (mbedTLS itself, AES encryption methods, etc), something about 100K, but need to check more precisely.
Estimation of this task is about 40 hours with debug and testing.

ralcaide, 2018-04-03:
Ok, thank you for the feedback. By the moment we are not sure about how to proceed. 

@juanra-bascones , let us know the result of your experiment in order to have more info to take a decission.

juanra-bascones, 2018-04-03:
@ralcaide :  

Is over there ANY https server using the very same certificate we are using for messaging to the broker?
I so... I would like to use it to try to make the test..

Doesn't matter what the server does, I just will try to get the default page by https.
Even best, if it would had a web server with access to the same content (any), by http and https (in two different ports, of course), and using the same Certificate I have (or other that you could provide me).
I am pretty confident that simply using the (already existing) TLS flow with the HTTP commands is going to work fine, but we need to verify it...
Once verified, the #31 should be very close to be done!!