Sinapse-Energia/bootloader-stm32 #32: [Cortex M3 - GPRS BG96] The bootloader is not working with BG96

ralcaide, 2018-06-14:
The bootloader is not working with the transceiver BG96. Understand the problem & fix it

juanra-bascones, 2018-06-20:
Introduction
After a preliminary view, the first reason for the Bootloader not working with the Quectel BG96 modem was very obvious: in the initialization phase, even when the transceiver is correctly recognized, the chosen commandflow used to connect to the server was hardcodad to the M95 one... so there was pretty understandable the fail in the conenction preocess.
Fixed this, there is still a much more complex and sophisticated problem, hidden until now by the former flaw.
The fact is that, after fixing this first flaw, allowing the program to open a sucessfull TCP connection to the server, and the HTTP GET command intended to download the binary file that is sent over the connection, the file got by the download loop doesn´t  pass the checks to verify the correctness and integrity. More specifically, because the file was partially truncated.

juanra-bascones, 2018-06-20:
In the search of the bottleneck/weakness..
In order to isolate where, within the whole end to end download path (web server, GPRS modem, uart, temporal buffer, main program loop..) could the data be missed out, several test have been carried out.
After those test is seems quite proven that:
1) The reading pace the main loop does over the temporal, circular buffer, seems to be fast enoug (the test revealed that the main of the times there is no more than a few bites available to be read), so the overrun of the buffer doesn´t seems to be the cause of the data missing.
2) Several alternate ways of populating the temporal buffer (irq handlers, DMA..) and several implementation of the circular buffer has been put under test, and none of them have solved the data misisng.

As the understanding betwwen the Quectel BG96 Modem and the web server to deliver, over TCP and GPRS, every char of the download is definitelly out of the debate, the only segment under suspect seems to be the one betwwn the GPRS modem and the interrupt populating the temporal buffer, and this conclussion puts the UART just under the focus.

juanra-bascones, 2018-08-09:
The files for the BG96 modem, and the (few) changes made in Tranceiver and utils files to support this modem has been integrated in the M3_unification branch.
It has been tested with both servers (sinapseenergia & eluxoon) with results apparently better for the second one. 
Intesive testing hasn´t been possible because a problem with my BG96 Board, that make testing quite difficult. 
More intensive testing will be made when the problem get solved.