Sinapse-Energia/bootloader-stm32 #33: [Refactoring] To explain the improvements to be done

ralcaide, 2018-06-25:
The code of the bootloader is not optimal and should be improved. Write here the proposed improvements:

juanra-bascones, 2018-06-26:
In fact, the Bootloader akss for several improvements that can be (some of them, must be) done.
This is a concise summary to just enumerate them:
(1) The eventual truncation problem (see #   issue)
(2) The non consistent use of the (potentially not valid) downloaded file to start the application
(3) The use of interrupt callback method instead of dma 
(4) Only working on TCP connection, not having never used TLS connection
(5)  A "mestize" nature, where not one but two different circular buffers are used.
(6) Use of "ligthly modified" transceiver files (base, and derived classes..) and utils.c files, making harder the bootloader evolution.
(7)  Main loop behaviour (where, right now) downloading a typical file of 100-150Kbytes is using several millions loop iterations and +100.000 one or two bytes flash writings   
(8) Much less than optimal way of calculating crc for integrity, where the stored flash downloaded file is read byte by byte (so, +100.000 flash reading operations).

(1) and (2) are real, very serious, bugs causing the bootloader to have sometimes a inappropriate behaviour, so there need to be addressed with the maximum priority
(3) has not been a problem so far, but has a big potential danger underhead, so it must be addresse asap, also.
(4) is just a matter of functional coverage, and, first of all, some tests have to be done to identify if how much code (if any) has to be changed.
(5) and (6) are, indeed, consequences of  former integrations already done, where the avavilability of a working result were prioritary, even paying the price of having "some loose ends" like those.
(7) and (8) are design improvements to be taken into account (trying to get a better performance and, spacially,  a more reasonable use of the NVM accesses).

In following comments, we will analyze those points deepely.

ralcaide, 2018-06-26:
Ok, as always, we need an stable version and then improve it. I think we should focus on 1 and 2 and then to plan tasks in order to improve the other points