Sinapse-Energia/bootloader-stm32 #35: [Cortex M3] The UART is not able to work in different speeds

ralcaide, 2018-06-26:
After perform some testing seems that MCU UART works correctly only with 115200 baudios (Only speed valid to perform connections). We need to increase /decrease the speed in some cases.
Identify why is not possible to work with a different speed

ralcaide, 2018-06-26:
@fpgamcu , the problem is related with other product with M3 MCU but the base code is the same as M4. Could you please test if in Cortex M4 branch we can work with different UART speed?

Then, could you help us to identify where is the problem in M3 branch, please?

ralcaide, 2018-07-10:
@fpgamcu , could you please provide us feedback about that?. Any idea?

fpgamcu, 2018-07-23:
How was you changed the baud rate in M3 code?
Using [Socket_bank.c line 90](https://github.com/Sinapse-Energia/bootloader-stm32/blob/ab5d3b111b6c813ef971be6c250ff20d2a5e341f/Src/Socket_bank.c#L90) or [Transceiver.cpp line 545](https://github.com/Sinapse-Energia/bootloader-stm32/blob/ab5d3b111b6c813ef971be6c250ff20d2a5e341f/Src/Transceiver.cpp#L545) ?