Sinapse-Energia/bootloader-stm32 #36: [Cortex M3 - ALL] To decrease the size of the executable

ralcaide, 2018-06-29:
This issue comes from #34 
The bootloader needs more than 48 KBs but its size in memory is limited to 48KBs. 
It is necessary to decrease the size following different strategies:

1. To compile in release mode instead debug and maybe use some optimization options
2. To delete dead code
3. To use several #Define in order to compile one part of the code of another part, for example, M95 vs BG96 vs HLK

juanra-bascones, 2018-06-29:
After not having the expected results in the #34 issue, we have focused in the more short time part of this one, that is, the use of Release configuration, instead of debug configurations, to see and compare results, both in code size, and in behaviour.

_Fisrt of all, generate under Release configuration, hasn´t been as straigthforward as it should be. At least in my development environement, the Release configuration for this project had never been used, and it didn´t even compile._

Just in case this could be the situation for other's IDE, I am going to summarize the actions I had to do to get it ready:

I had to adjust the directories set for Release configuration 
(Properties->C/C++ General->Paths and Symbols) for Release configuation, copying (one by one, the IDE doesn´t have an option to copy all at once) all the Debug entries, in both C and C++ languages.

Once having it compiled, I got a good new : the code segment (the one which matters) got reduced from 52504 bytes of Debug to 39892 for Release (it is arounf a splendid 25% of rebate.. a figure we can take as a first approach of what we can get in general ..).

For making possible to run the Release result from the ST-Link probe, it is necesary to edit the debug configuration making "Release/bootloader-stm32.elf"    the  C/C++ application to be run (if not, it will try to use the one in the Debug directory, if it exist).

For testing, we have to start the debugger as usual (pressing the bug icon, for instance)... 
CAUTION: After writing the flash, as allways, don´t forget the IDE expecting the program to run at once.... The Debug stops (as allways) in the very first line of our program (but without showing the source code, because there is no source line info).
Just press Resume (or F8) to let it to run again (now, indeed, definitely..).

If the compilation have been made keeping the DEBUG pragma, the led will give the us a visual feedback of the bootloader way (the led flashing in yelow during the connection and in magenta during the download...).
The Bootloader completes the download and succes in booting the client application.
Not only the very first time, but also in the sucessive ones, where, just updated, the BL yields inmediate control to the application.

ralcaide, 2018-07-10:
It is not necessary to decrease the size. It is only needed to compile in Release mode.

closed 2018-07-10 by ralcaide