Sinapse-Energia/bootloader-stm32 #37: [Cortext M3 - M95-BG96] To decrease size of bootloader deleting BG96 Code

ralcaide, 2018-06-29:
This issue comes from #34.
At this moment the BG96 is not working, so delete this code in order to get a smaller bootloader in order to be valid for M95. 
This is a transitory solution before finish with #36

juanra-bascones, 2018-06-29:
Surprisingly, **removing all the BG96 code doesn´t reduce de executable size in a significative way** (in a first cut, I doesn´t seem to reduce it at all..).
We have to review this test in order to better understand it, but, in fact, the most of the waste for the connection methods ( BG96, M95, TCP and TLS...) is in RAM memory (the structure arrays of the connection steps, the char arrays to build-up the AT commands with variable parts, etc..), but the code to implement them is minimum, because are msotly a few calls to base class methods.

**What it did result to have a quite important impact in executable size is to compile the project in Realease  (instead of Debug) configuration.** 

In order to keep fidelity to the issues target, we propose to keep this issue in stand-by (it matters, but it doesn´t seem to deliver the short time result we did expect) , and move the discussion to the inmediate efforts to the #36, which become now the prioritarie one.

ralcaide, 2018-06-29:
Ok, ON HOLD