Sinapse-Energia/bootloader-stm32 #38: [Cortex M3 - HLK] To integrate HLK transceiver in the bootloader

ralcaide, 2018-07-10:
It is necessary to integrate the HLK transceiver in the bootloader for Cortex M3.

juanra-bascones, 2018-07-14:
I did created a specific branch from the Cortex M3 master branch, with the purpose of having a version of the Bootloader able to work with the Hilink HLKRM08 ethernet modem.
As alredy been stated in the #33 issue, the bootloader exhibist quite a few weakeness, and it is very advisable to address those items before going to any ulteriro evolution. 
Nevertheless, as the goal for this issue is to get the most direct path to the result, I did take the approach to "change less things as possible".

There are good news and bad news.

-  The good news:  there already is a version of the Bootloader which, by mean  of selective compilation, can work with Hilink RM08, or with Quectel M95.

- The bad news : the downloaded file is not allways complete and valid. Sometimes happens a very similar truncation to the one with the BG96.

So, **it seems we have the very same problem with both devices** (RM08 and BG96), and, for the time being, the hypothesys we work with is that the UART is overwhelmed by the modem speed.

(**IMPORTANT**!!! : even when the download is complete, without truncation, and passes the crc check,. etc..), the final step, calling the Boot_StartApplication() function never has worked out. (because the binaries we have for downloading are intended to work with M95/BG96 modems, and when they try to start, they never get started... nor connected nor even properly initialized..).
So, in order to be able to run some end to end tests it´s necesary to have in the server some files compiled for the very same board we have running the test.

juanra-bascones, 2018-07-15:
From a global point of view, it doesn´t seem that the integration done has been succh a a good deal..
The work for having it all, up and running, has been quite big.. much more than expected, and with a very important load of debugging a and bug investigation...and the result doesn´t seem to deserve the effort: the code is clumsy, confussing and poorly legible in many aspects.

The inmediate recommendation is to schedule a major rewrite of the bootloader (and also, in part, of the M2M code), in order to factor out the equivalent, but misplaced code (specially initialization, and Hw related functions) to a separate file, where the integration can be tracked in a productive way.
This recommendation is another one to be added to the improvement list recollected in # 33.  
There are, also, some very particular items to think about.
For tecnical reasons in the M2M integration, the RM08 imposes the use of the MPU TealTimeClock, which has a important impact in several initialization functions, and also imposes a foorprint in terms of several new system files having to be added to the bootloader project. Maybe the need of having a reliable datetime base is no so important in the case of the bootloader, and, in case of memory shortage, this item need to be reviewed in a future.

juanra-bascones, 2018-07-15:
At the current branch state, the same code compiles (setting the correct tags in definitions.h) for M95 (#define BUILD_M95  & #define DEFAULTBOARD Board0), and for RM08 (#define  BUILF_RM08  & #define DEFAULTBOARD Board2). 

The first one works fine until client start, connection and registration, and the second seems to work (with the eventual truncation problem and withown having the fianl feedback of the application flashing and tracing, as we said in previous comments).

The code for BG96 handling remains off line, as fas as the changes to supports are still in another side branch, waiting for a solution to the truncation problem.

As far a suitable binaries for RM08 were availables for download, more test will be driven, and alsoi for the other Base board using RM08, which haven´t be tested so far.

ralcaide, 2018-07-15:
Thank you Juanra for your great job, as always. Yes, you are right, we need to improve the bootloader asap in order to have a maintainable code. I will plan the tasks for the next sprints.

Also, tomorrow I will generate the needed binaries for testing with HLK.

juanra-bascones, 2018-07-16:
To keep in view the margin we have under the maximum bootloader available space, here are some figures for the current state of the branch
- Compiled w/Release BULID_M95 : Code 40652	RAM 616	   consts 3808  
- Compiled w/Release RM_08:          Code 39908	RAM 624	   consts 380

juanra-bascones, 2018-08-07:
The current state of the main branch have been tested with both Quectel M95 and Hilink HLKRM08 modems (the last in both Ethernet and Wifi modes), and works fine.

After applying a few preprocessor labels for the selective compilation (mainly, but not only, for the modem to use), the size of the executable is again under the 48 Kb limit, even for the Debug configuration (which are good news for the development process).
 
The test have been run using the client applications B0-ALL-ETH-JR-V1.26.3.bin and B0-ALL-ETH-JR-V1.26.3.bin. These are "all modem/all boards" M2M application, with the only difference of the ethmode variable value (as far as this variable is defined in compile time and there isn't still any way to change it afterwards, there is needed a separate executable to test each one of the transport mode values). The executable leads himself to recognize the board by means of the value of a NVM vayable (BOARD), so an previous execution of the application in local mode is perceptive to store the rigth value of the variable into the NVM.

It seems that the only "make-sense" combinations to be tested are the ones using the very same mode in the Bootloader and in the client application (eth mode BL compilations downloading the B0-ALL-ETH client, and the wifi mode BL compilation downloading the B0-ALL-WIFI.
Nevertheless, hybrid combinations have also been tested, and although the one "eth BL downloading Wifi client" works rigth, the converse "wifi BL downloading eth client" doesn´t (the client downloads but never get connected). 
Although these hybrid modes are never going to be used, this topic need to be clarified because it seems there are some loose ends with the initialization/deinitialization processes.

juanra-bascones, 2018-08-07:
The hardest and most time consuming part of the testing has been the rigth selection of the application client to be used, along with the rigth configuration of the NVM board to make possible the identification process.

It is very advisable to review the complete way to get the downloadable application in order to include the suitable checks/verifications in order to verify and identify the correctness of the binary file

juanra-bascones, 2018-08-08:
The current state of the master (Cortex M3) branch has been tested in a wide variety of combinations (two web servers, wifi & ethernet layers, two transceivers, and over the three base boards under test).
The executable size is still small enogh to be able to run the tests in debug configuration.
In all cases the test have been run:
1) making the ad-hoc compilation of the BL for the set of options (web server, client file , modem, board, transport..)
2) Having the board to identify itself, by mean of the "BOARD" variable value in NVM, put there by a previous execution of a IDE client

We take this approach because for the Board1 board, the execution of the files B1-ALL-XXX-JR-V1.26.3.bin, from any server, didn´t get properly connected when the erased chip.

Investiating deeer, we found that the only board where those files get properly run was the Board0 (the one used for Quectel M95 modem), what seems to suggest that,  that, in spite of their name, the files have been compiled with the Board0 as DEFAULTBOARD.

As a summary, 
- The BL version seems to work fine over all the known configurations (pending to solve the still open issue with BG96).
- The whole chain to get the binary file for bootloader download demonstrates not to be mature enough, and deserves to be reviewed in a near term.