Sinapse-Energia/bootloader-stm32 #40: [Cortex M3] Makefile for Bootloader

ralcaide, 2018-08-30:
Apply same methodology of M2M client to BL

juanra-bascones, 2018-09-04:
The list of build parameters for the Botloader will be:

* BOARD
* TRANSCEIVER
* APN 
* ETH_MODE / SSID / PASSWORD
* HTTP SERVER
* HTTP PORT
* BINARY PATH
* BINARY NAME
* UPDATE_FW
* UPDATE_FW_COUNT

Is there anyone missing or anyone left?

ralcaide, 2018-09-04:
I think these are the necessary parameters. Go further with that ;)

ralcaide, 2018-09-14:
Testing for DDP (IoT Presence)

ralcaide, 2018-09-14:
@juanra-bascones , it is necessary to include also in the bootloader project the last improvements done in M2M: https://github.com/Sinapse-Energia/M2M-CommBoard/issues/128#issuecomment-419629394

juanra-bascones, 2018-09-17:
The make-enabled version for the BL y ready to testing.
The use is the same one that the M2M one (each one with its own set of optional parameters, of course).
But the main options are the very same (ROOT, OUT, PFILE...) with the same dafault values for both.
The help pseudotarget is also implemented in the BL, with its particular option list, of course

ralcaide, 2018-09-17:
Thanks Juanra, it will be tested and feedback provided