Sinapse-Energia/bootloader-stm32 #43: [Cortex - M3] To adapt the BL to new behaviour of HLK and take into account the retries

ralcaide, 2019-06-28:
The HLK driver have been updated in the client side of the program in order to manage the configuration in a easy and sustainable way. Here the linked task: https://github.com/Sinapse-Energia/M2M-CommBoard/issues/155

It is necessary to modify the BL to work with this behaviour in a secure way, allowing the reset if the HLK if the BL is not able to connect to internet.

juanra-bascones, 2019-07-01:
Fully agree.  Indeed, the Bootloader not only deserves the same the self-service configuration capabilities being now in the client application, but probably it deserves them even more. 

The point is that the BL is, quite simpler than the client app from many aspects, but more complex from others.  And this is probably one of those where it's a little more complex.

We will try to summarize the situation in the next comments.

juanra-bascones, 2019-07-01:
### Current situation

The Bootloader has already a certain retries policy, where, if it gets a download failure, retries for a certain number or times.
It is implemented by a quite simple retry loop like it shows this code excerpt 

```

        uint8_t attempt = 0;
	while (Boot_PerformFirmwareUpdate() != BOOT_OK) {
		attempt+=1;
		if(attempt >= NUMBER_RETRIES) break;
		HAL_Delay(5000);
	}
	// .. some deinitialization stuff
	Boot_StartApplication();

```
The bootloader calls the Boot_PerformFirmwareUpdate() function (which connects to the server, downloads the binary and checks it)  until it succeds, for a maximum of NUMBER_RETRIES times..
When it succeds (or after NUMBER_RETRIES failures)  drives to Boot_StartApplication() funtion (that gives control to what is in NVM, whatever it were).

Until now,  NUMBER_RETRIES has been 1, so, the behaviour is to try download once, and, no matter what happens, execute what were in Flash memory program adress.

This behaviour is partially satisfactory:
- If the download goes fine, the client is updated and runs (OK)
- If the download fails, and there is any client version in Flash, it will stiil runs (NOT fully OK.. simply demores the update for later on) 
- If the downloads fails, and there isn´t any previous client version in Flash, it eventually ends in a HardFault, and the Bootloader will take control again by means of the WDT. (clearly Inadecuate)

The situation is quite the same for other values of NUMBER_RETRIES...

juanra-bascones, 2019-07-01:
Why BL is different of client application?

When analyzing what to do when the BL whole process fails, we have to take in mind TWO  inputs  (without equivalence in the equivalent client start process): 

1) First,  and more important, we have to distinguish the reasons the BL can fail in its Boot_PerformFirmwareUpdate() purpose.
- "Recoverable"  (like a coverage problem, for instance) or "non recoverable" (for instance an invalid value of the SSID/password in wifi connections, or an invalid APN in GPRS connections, or an invalid url/port for the server) configuration problems. In both cases there is no downloaded data. 
- An invalid/incomplete/corrupted file download.
 (this cause, very far of being a extreme niche, has been the most important cause of BL failures in the past)

2) Also important, we have to differenciate if there is (or not) a former client version stored in the device Flash. It there is one, in some circumstances, it could be adecuate to yield to it (at least after recurrent problems in downloading a newer one). It there isn´t any, it simply couldn' t be an option

So we can begin to build a two dimensional table where all those circumstances can be combined, and then decide what to do in each particualr case.

A proposal can be the following one:

Download result  |    Former client  |  No former client
-------------------| ---------------- | --------------------- |
  |    OK  |          BOOT | BOOT  |
 | No download   |  Retry/BOOT  |  Retry
| Invalid downloaded data  |  Retry  | Retry

juanra-bascones, 2019-07-01:
In order to implement this table contens (or another alternatives that we can debate...) we will need:
- To have the way to certainly distinguish where there is no data at all, and where there is data but is not valid (we need to verify the Boot_PerformFirmwareUpdate() return values in order to certainly check their meanings)
- To know if there is /isn´t a preexistent client version in Flash.
It seems that the NVM stored Context variables FWNAME and FWVERSION are going to be what we need. If there is a client in Flash, its name and version will be stored in NVM. If there isn´t none, they will not be. The only problem is that the BL is not aware of this variable so far. It retrieves some variables from NVM, but the name/version of the client are not among them.

So there are TWO tasks we have to work in as soon as possible:
1) To check the fully correctness of the Boot_PerformFirmwareUpdate() return value, and correct them if necessary
2) To wide the CLIENT_VARIABLE used for the main Bootloader process, to make room for the client name and version.

juanra-bascones, 2019-07-02:
Yesterday I began to work on all this...but I did found the situation is a little worse that I expected..

As we have noticed in the past,  the M2M and BL projects are very far away from each other. (former "integrations" from M2M changes to BL did in fact result really painfull because the very different phyical structures of both projects)
**The first desappointing discovery now is that the retries loop simple doesn´t work just setting the number of retries bigger tan one.** 
It doesn´t work because one of the big differences between both projects: the communication layer.
BL uses some M2M inherated infrastructure (the command flows, the circular buffer, the modem classes, etc..) ONLY  during the connection phase. When connected, an "evil" global variable changes its value to keep using the "tradictional way" the BL did from the beginning, and it is far apart from all the M2M framework. 
This global variable (called  application_layer_connection) acts like a kind of semaphor to use one/another thing: the variable is initializated to 0, driving to the M2M inherited code (Modem methods, circular buffe, etc...) and when the connection phase completes, its value is changet to 1, changing the destination for the UART interrupts, and giving control to the old code for the soket writing (the http get command) and sochet reading (by an array and a set of variables).

So when the download succeds, client application starts at this way from scratch and there is no problem. But if the download fails and there is going to be a new attemp, the flow simply doesnt work because the Transceiver class and its buffer are functionally dead, and any call to its methods fails.

Maybe the simplest (and dirty) way to solve this particular problem is to change back the global variable application_layer_connection, to set it again to 0 when the download fails...
I will try it today, and maybe this will solve this problem..
But we can find more of these around the corner....
and looking for simple (and dirty) solutions for them is going to let the code even worse and even more confuse...that it already is

A year  (https://github.com/Sinapse-Energia/bootloader-stm32/issues/33#issuecomment-400243190)  I did propose to schedule for a major rewriting of the BL in order to solve  its main maintenance problem: the big differences with the M2M framework and the strong difficulties to integrate the M2M improvements into the BL code...
Maybe if this rewriting have been done in the meanwhile, this issue topic work would be a snap....but in fact it's not.

**Can be feasible to schedule this rewriting , to fully align the code of both projects (in order to them to really share a set of common files and to do the commom tasks in excatly the same way with the very same code)  and postpone  this (or other BL evolutive issues) until this two  project convergence?**

ralcaide, 2019-07-02:
Thanks for the great analysis Juanra, as always. Let me check what we have right now on the table and I will provide you an answer.

If yes, what could be a realistic estimation?

juanra-bascones, 2019-07-02:
I will analyze the code with more detail in order to have an estimate as soon as possible.

juanra-bascones, 2019-07-03:
After a detailed review of the current Bootloader code, the conclusion is that there are enough things to be changed to get the ideal target we would like to get.
We will start by the summary of both, the current state and the proposed goal to achieve in the following comments.

juanra-bascones, 2019-07-03:
### The Bootloader now
The bootloader project consist of the following files
- BootLoader.c:  This is the "core" code, where the BL behaviour  is implemented (specially its starring function Boot_PerformFirmwareUpdate().
- Socket_bank.c : This file implements a set of socket-style functions (Socket_Init(),  Socket_Create(), Socket_Connect(), et...) to be used by Boot_PerformFirmwareUpdate(). 
Some of those functions (like Socket_Init() and Socket_Connect(), for instance) are mainly front-ends to old versions of M2M communication functions (that have been seldomly updated, and are still in old fashioned versions).
Other functions (like Socket_Write() and  Socket_Read()) have their own vertical and self-contained implementation, using for that, the old buffer and associated variables and functions to a very primitive version of the M2M communication infrastructure (from M95Lite.c).
- M95.c, BG95.c RM95.c Transceiver.c circular.c utils.c  are out-of-date versions (and partially modified in many cases) of its counterparts files in M2M project.
-Crc32.c and Flash_NVM.c, contain low level functions needed for the BL logic. Crc32 has nothing equivalent in M2M, but an important part of Flash_NVM file contents is very similar to the equivalent M2M functions in southbound_ec.c

In summary, the BL consist of
1) Functions 100% BL specific, whitout no M2M equivalence (like Boot_PerformFirmwareUpdate or all the CRC functions)
2) A lot of code (like all the transceiver files, utils.c or circular.c), borrowed from M2M in some moment in the past, and later modified to fit BL needs.
3) Some code (basically all the Socket_bank stuff) implementing, by its own way, operations (like Send and Get data) that M2M implements with very similar purpose, but in a different way.
4)  Finally, main.c file has some content "similar" to some other in M2M main.c (like the initialization calls for Uarts, SysmtemClock, or Timers..) with other BL specific like recovering the shared NVM variables and implement the heuristic for downloading or not the client file.
 
Needless to say, that in current situation, any "_integration_" process (i.e. traspose some functionality already developed and testsed in M2M, to the Bootloader) is an **authentic pain of neck**.

juanra-bascones, 2019-07-03:
### The Target
In very few words, the target we really would like to reach is to factorize the common, shared functionality of both projects (M2M & BL), into a unique set of shared and unmodified files. 
It's kown that our runtime environement doesn´t allow to have DLLs (or shared libraries), but the intended goal has to be to keep the same sharing level and discipline... but with compilable source code instead : _the code is expected to work for me at once, but if it doesn´t,  I will ask for the proper update, but I will never modify it_ (in the same manner that I can not modify the DLL or .su files, by obvious reasons).

To get this goal takes quite a lot or code changing, rewiting and deleting...but, aditionally requires to have clearly identified the stable M2M code we want to use to integrate with. And here we face another side problem:  we have now at least 3 or 4 evolutive branches in M2M  (the ones for **BC96 NB-IoT**, **STPM10**, **SPI-RFID-LCD**, **HLK-WebConfig**, as long as I can now remember..), spawned all of them from F2_unification "master-like" branch, and with important concurrent modifications in many of the files needed to be shared with BL.

Moreover, the physical and logical structure of the M2M code have _degenerated_ over the time: there are now like 30 files in Src folder and  even a few more than this in Inc folder, having _swimming together_, the core files (like parametrization.cpp or application.cpp files) with the ones for the framework (Context, Frames, Timeevents) and with other optional or product specific files  (like the ones for CAN or I2C buses..).
The only "exceptions" to this mess are the two distincted Libraries (Paho and SunSet). 
A very important improvement I have in mind is to "packetize" all this stuff in more distincted libraries, with specific criteria (like Buses Libraries, Transceiver Libraries, etc...), and **I highly recommend to create an issue for that**.  Thinking in having this packetized future version of the M2M code, the main goal for the M2M-to-BL stable relationship would be to define the packages they have to share and hereinafter, to evolve them allways having both projects in mind and driving regression tests for both before commiting any change in those common Libraries.

ralcaide, 2019-07-04:
Thanks for the feedback Juanra. 
I see we have a problem with the refactoring of the code. We should plan these tasks asap but in a smart way, starting with the code shared between BL and CL. 

I've created a global task here: https://github.com/Sinapse-Energia/M2M-CommBoard/issues/159

Let's speak and plan the subtasks together.

juanra-bascones, 2019-07-04:
In the next comment (what I'm working on) I'm trying to decompose all the tasks involved in this evolution proposal, in a organized way, identifying several sequentializable stages, giving for each of them a first guess of complexity and the practical benefits we can have with each one.
I hope to have it done along this morning
If you want, we could have a speech about it when you can read it.

juanra-bascones, 2019-07-04:
### Roadmap proposal
Now, back in the "main thread", of how to face the BL update, and beacuse the important scope it has, I propose to go there by a few sucesive stages, being the last one of them dependent of the time the M2M project could have all their side branches integrated (and, if possible,  packaged in a wise way).
Here it comes the stages relation:

1) In my opinion, the very first stage has to be to definitelly remove all the "collision" or "competition" code in BL (i.e, the one in the third point of the "The Bootloader now", last paragraph list). This means to replace the primitive BL-specific communication structure (main part of socket_bank.c file) by the code alrady existing in M2M Framework. Its possible (near 100% odds) that these replacement could imply make some modifications to the existing M2M code (for instance, the existing GetData() method, messag polling oriented, is based in "waiting" a little to the data to arrive, while the BL http reading is mainly concerned with avoid overruns and has to try to pull data as fast as possible.
Also the current GetData() method behavoiur is of  "give me the N bytes I need or don´t give me nothing", because the Mqtt message structure is a 3-step process where each chunck size is very well known beforehand... and only is usefull if is complete, while the http read is "give until no more data seems to rest". 
These differences probably will impose some changes to be done in existing code.
In summary, for this first step, the goal has to be to have: 
-->**All the "common purpose code" as similar as possible** (this means to be alble to use  some of the files unchanged and some other files, sligthly modified..)
-->**All the BL specific code only based in the previous common (or almost-common) code**.

2) Once all the legacy code had been removed and replaced by their equivalents from the M2M framework, the next step could be to address the shortcommings, weaknesses, restrictions, etc.. we have already indentified in BL code (see..), as, for instance are:
- The use of DMA instead of IRQ handlers (at least, for Quectel modems).
- The use of TLS as transport layer to be able to use http and https servers
- The already diagnosed, but not fully solved, truncation problem with BG96 and RM08 modems
- The extremelly unefficient use of NVM during download. A mediem sized 150K client file can drive near of 100k  NVM writes (medium size write is 1 or 2 bytes), an even more reads (the CRC verification reads the downloaded file byte by byte!!)
- **The correct implementation of retries according with the already explained in former comments in this issue**

(maybe, some of those improvements could induce aditional changes in some framework files)

3) Next stage has to consist in "zoom out" and identify the reasons of all these file modifications, and foresee  "fussion versions" for all of them, in order to be shared unchanged by both projects (with the minimum use of "preporcessor sandwiches"...if possible, nothing at all). 

4) Once any of these "discordance files" could be reconciled in a unified version,  is very important to follow a strong discipline in its maintenance and evolution, to keep the unified version along the time. 
In order to make this goal easier to follow, is very important to "packetize" these demonstratedly saherd files in **libraries** or **packages**, to be used in M2M and BL (and perhaps, in other projects existing or futures), but never modify them as part of the project maintenance  (as it happens now with Paho library, for example).
This stage has to be drawn in both projects (BL and M2M) at the same time, and much more deeply in the last one because the number of files is much higher, the reasons for the code to be packed .

juanra-bascones, 2019-07-04:
### Overall Estimation

1) For the first stage, the work to be done is fully technical. 
There is pretty much code that has to be droped out and replaced by new one, supposedly equivalent, and there implies quite a lot of testing and debugging.
It's not so easy to estimate the work load, but a loose interval can be to think in **20 to 40 hours of work** (maybe more... if unexpected problems will emerge, let's hope not..).

2) After having the first stage deliverable ready (a "transition" release of the BL, where all the ovelaping code has been definitelly removed), it will be possible to schedule any (_none_, _some_, or _all of them_) of the "hot spots" related in stage 2 previous comment relation.
**Each one of these, has a different complexity** : for instance, setting up for using DMA with UART6 (GPRS) modems, should be **quite simple**, and using TLS **a little more** but not very much (both are features already working in M2M). 
In particular, **rigth working of retries mechanism** (and its associated drving to the web config tool for the RM08 modem) **surely has to be very simple too.**

3) Stage 3 is mainly a sw generalization task, involving the reconciliation of sligthly different functions and/or classes in one set of general purpose functions / clases (whether by mean of extra parameter, state variables, or whichever could be needed , including, as a last resort, the use of preporcessor directives). 
The real effort of this task will depend of the number of functions/classes to be reconciled, but as a general overall estimate, we can guess **10 hours aprox.**

4) Finally, the stage 4 must leverages on the existence of a reference M2M unified version  (and, maybe, already partially packetized) of M2M project (whose effort is not considered here).
With this unified varsion, this stage task is to try to identify the extra packages where most part (ideally, all) of the common code between BL and M2M can be placed, define the right way to use these packages, and state the procedures to maintain and evolve these packages hereinafter. 
This stage is mostly engineering oriented, and its main task are going to be to agree balance between projects roadmaps and to create the guides to encourage compliance. **This could be a background task, to be fullfilled in collaboration, with marginal dedication over  some period of time**.

juanra-bascones, 2019-07-08:
I'm already working on this, with basically the same requirements and general target, but using the Hub_Home board and its baseline project (from https://github.com/Sinapse-Energia/IoT-Hub-Home/tree/IoTHubHome_Bootloader). I work on the same external PC I have usually using for recent developements.

By now I am charging the times to this same #43 issue, but maybe it would be better to have an issue by its own in the Hub_Home Bootloader project, not only for traceability and accounting, but also for have the issue information available to the potentially interested people from Hub_Home project universe.

ralcaide, 2019-07-08:
Sure, here you have the task: https://github.com/Sinapse-Energia/IoT-Hub-Home/issues/38